adafruit/Adafruit_VL53L0X #9: Long range mode

elpedriyo, 2018-04-08:
Hello,

I was just wondering how can I change the different modes of the sensor with this library.

Let me know,

Thanks

Gomezzzzz, 2018-05-11:
Hi, it's all I could find.
https://satoshinm.github.io/blog/171111_enabling_long_range_mode_on_a_vl53l0x_time_of_flight_sensor_in_arduino.html

austinBontragerRealmFive, 2020-06-03:
I would also like to know this! I don't understand how to actually accomplish this, based on that blog post.

RagnvaldIV, 2021-01-06:
I'm looking into this for a project and believe I found a solution. The `Adafruit_VL53L0X::VL53L0X_Sense_config_t` that is called in the vl53l0x_multi_extended example (line 32-33) caught my eye. 

Looking a bit further down in the vl53l0x_multi_extended example (starting line 47), we see the list of sensors. Here, they're setting the values as described in the sensorList_t struct for each of the 4 sensors. Then at line 94 they call `begin()` for the sensors in that list with the struct components as arguments. In the simple example, they call `begin(address)` but we can make use of the function with more arguments to set the mode.

Looking into the library files, I see that the config is an enum:
`typedef enum {
    VL53L0X_SENSE_DEFAULT = 0,
    VL53L0X_SENSE_LONG_RANGE,
    VL53L0X_SENSE_HIGH_SPEED,
    VL53L0X_SENSE_HIGH_ACCURACY
  } VL53L0X_Sense_config_t;`

Therefore, I create a variable to set my config: `Adafruit_VL53L0X::VL53L0X_Sense_config_t sensor_config = Adafruit_VL53L0X::VL53L0X_SENSE_LONG_RANGE;`

Then I call begin with the full args: `lox1.begin(LOX1_ADDRESS, false, &Wire, sensor_config)`

I haven't tested this yet (still waiting on the parts, thought I'd get ahead on the code), but it compiles and should be setting the mode. I hope this helps!

adaptandadopt, 2022-02-22:
> I'm looking into this for a project and believe I found a solution. The `Adafruit_VL53L0X::VL53L0X_Sense_config_t` that is called in the vl53l0x_multi_extended example (line 32-33) caught my eye. [...]

@RagnvaldIV , did this work for you?

adaptandadopt, 2022-02-25:
Well, I gave it a shot, and this worked for me:

```
Adafruit_VL53L0X lox = Adafruit_VL53L0X();
Adafruit_VL53L0X::VL53L0X_Sense_config_t long_range = Adafruit_VL53L0X::VL53L0X_SENSE_LONG_RANGE;
#define SENSOR1_WIRE Wire

TwoWire *i2c = &Wire;

  if (!lox.begin(0x29,false, i2c, long_range)) {
    Serial.println(F("Failed to boot VL53L0X"));
    while(1);
  }
```

caternuson, 2022-03-15:
Closing. I believe this was resolved with:
https://github.com/adafruit/Adafruit_VL53L0X/pull/43

The `vl53l0x_multi_extended` example mentioned is one good reference.

See also documentation here, can either set when calling `begin()` or later with `configSensor()`:
https://adafruit.github.io/Adafruit_VL53L0X/html/class_adafruit___v_l53_l0_x.html#a015bf7435cd17463460a943135900a87
https://adafruit.github.io/Adafruit_VL53L0X/html/class_adafruit___v_l53_l0_x.html#acc6de06624c431917af24888f487b082

closed 2022-03-15 by caternuson