adafruit/Adafruit_VL53L0X #17: How to enable fractional measuring?

canDry, 2018-12-24:
I see that the api has VL53L0X_SetRangeFractionEnable.
  
Has anyone got this enabled and working?  Plz share the code! 
  
It isn't enabled by default and so measure.RangeFractionalPart always returns 0.

caternuson, 2022-03-15:
It needs to be enabled first, otherwise the value will not be updated:
https://github.com/adafruit/Adafruit_VL53L0X/blob/9b0321580f830e65e07fd89469f1f9713f8aadd9/src/core/src/vl53l0x_api.cpp#L2283-L2290

`VL53L0X_SetRangeFractionEnable` is the underlying API call. But that is not currently exposed via the Adafruit wrapper.

The datasheet does not provide much information on this feature. What are the acceptable value for the `Enable` parameter of `VL53L0X_SetRangeFractionEnable()`? The feature seems to be a simple on/off toggle, so could just be single bit. But the `Enable` parameter is typed as a `uint8_t`, so unclear what all of the bits might potentially do.

I did a quick hack of setting this as `1` and it seems to at least start returning something:
[image]
Apparently the resolution is only 0.25mm, so there are the four raw values of `0`, `64`, `128`, and `192`.

Does enabling this affect the non-fractional part of the reading? As well as any other aspect of ranging?

Closing for now. Can reopen if more information can be found.

closed 2022-03-15 by caternuson

canDry, 2022-03-15:
Thanks for following up on this!