adafruit/Adafruit_VL53L0X #20: Compilation errors for example codes in Arduino IDE

dpackman0, 2019-04-12:
I tried to compile the sample code and got the following errors
Tried it with all 3 examples, all had similar issues.

11x 'error: stray '@' in program'
3x 'error: stray '\205' in program'
3x 'error: stray '\251' in  program'
3x 'error: 'Skip' does not name a type'
2x 'error: 'TwoWire' does not name a type'
error: 'TwoWire' was not declared in this scope
8x error: 'TwoWire' has not been declared
error: 'i2c' was not declared in this scope
2x error: expected unqualified-id before numeric constant
error: 'Skip' does not name a type
error: 'Wire' was not declared in this scope
2x error: 'lox1' was not declared in this scope
error: call to 'boolean Adafruit_VL53L0X::begin(uint8_t, boolean, int*)' uses the default argument for parameter 3, which is not yet defined

In file included from [...]\Arduino\libraries\Adafruit_VL53L0X\examples\vl53l0x_dual\vl53l0x_dual.ino:1:0:

[...]\Arduino\libraries\Adafruit_VL53L0X\src/Adafruit_VL53L0X.h:39:1: error: stray '@' in program

 @chetan-prime chetan-prime Add support for specifying an alternate I2C bus other than the hardco… 1563a9e on Sep 29, 2018

 ^

[...]\Arduino\libraries\Adafruit_VL53L0X\src/Adafruit_VL53L0X.h:39:1: error: stray '\205' in program

[...]\Arduino\libraries\Adafruit_VL53L0X\src/Adafruit_VL53L0X.h:40:1: error: stray '@' in program

 @ladyada

 ^

[...]\Arduino\libraries\Adafruit_VL53L0X\src/Adafruit_VL53L0X.h:41:1: error: stray '@' in program

 @deanm1278

 ^

[...]\Arduino\libraries\Adafruit_VL53L0X\src/Adafruit_VL53L0X.h:42:1: error: stray '@' in program

 @ServerTech

 ^

[...]\Arduino\libraries\Adafruit_VL53L0X\src/Adafruit_VL53L0X.h:43:1: error: stray '@' in program

 @chetan-prime

 ^

In file included from [...]\Arduino\libraries\Adafruit_VL53L0X\src/vl53l0x_api_strings.h:33:0,

                 from [...]\Arduino\libraries\Adafruit_VL53L0X\src/vl53l0x_api.h:32,

                 from [...]\Arduino\libraries\Adafruit_VL53L0X\src/Adafruit_VL53L0X.h:69,

                 from [...]\Arduino\libraries\Adafruit_VL53L0X\examples\vl53l0x_dual\vl53l0x_dual.ino:1:

[...]\Arduino\libraries\Adafruit_VL53L0X\src/vl53l0x_platform.h:39:1: error: stray '@' in program

 @chetan-prime chetan-prime Add support for specifying an alternate I2C bus other than the hardco… 1563a9e on Sep 29, 2018

 ^

[...]\Arduino\libraries\Adafruit_VL53L0X\src/vl53l0x_platform.h:39:1: error: stray '\205' in program

[...]\Arduino\libraries\Adafruit_VL53L0X\src/vl53l0x_platform.h:40:1: error: stray '@' in program

 @ladyada
[... 226 lines cut ...]

[...]\Arduino\libraries\Adafruit_VL53L0X\examples\vl53l0x_dual\vl53l0x_dual.ino: In function 'void setID()':

vl53l0x_dual:42: error: 'lox1' was not declared in this scope

   if(!lox1.begin(LOX1_ADDRESS)) {

       ^

vl53l0x_dual:53: error: call to 'boolean Adafruit_VL53L0X::begin(uint8_t, boolean, int*)' uses the default argument for parameter 3, which is not yet defined

   if(!lox2.begin(LOX2_ADDRESS)) {

                              ^

[...]\Arduino\libraries\Adafruit_VL53L0X\examples\vl53l0x_dual\vl53l0x_dual.ino: In function 'void read_dual_sensors()':

vl53l0x_dual:61: error: 'lox1' was not declared in this scope

   lox1.rangingTest(&measure1, false); // pass in 'true' to get debug data printout!

   ^

exit status 1
'lox1' was not declared in this scope

chetan-prime, 2019-04-12:
I'm not sure where you got this .ino file from, but why would it have my handle and @ladyada 's handle inside it?
The .ino looks like a bad copy paste job from your part.

closed 2019-04-12 by ladyada

dpackman0, 2019-04-12:
> 
> 
> I'm not sure where you got this .ino file from, but why would it have my handle and @ladyada 's handle inside it?
> The .ino looks like a bad copy paste job from your part.

No need to be rude. I did not copy/paste anything. I simply loaded the library from the Arduino database within the IDE, scrolled down to example sketches and tried to compile. I'm assuming all of those remarks are somewhere within the included .h file(s) or maybe from the STMicro API files included in the Adafruit_VL53l0x library.

This is the code that was compiled
`#include "Adafruit_VL53L0X.h"

// address we will assign if dual sensor is present
#define LOX1_ADDRESS 0x30
#define LOX2_ADDRESS 0x31

// set the pins to shutdown
#define SHT_LOX1 7
#define SHT_LOX2 6

// objects for the vl53l0x
Adafruit_VL53L0X lox1 = Adafruit_VL53L0X();
Adafruit_VL53L0X lox2 = Adafruit_VL53L0X();

// this holds the measurement
VL53L0X_RangingMeasurementData_t measure1;
VL53L0X_RangingMeasurementData_t measure2;

/*
    Reset all sensors by setting all of their XSHUT pins low for delay(10), then set all XSHUT high to bring out of reset
    Keep sensor #1 awake by keeping XSHUT pin high
    Put all other sensors into shutdown by pulling XSHUT pins low
    Initialize sensor #1 with lox.begin(new_i2c_address) Pick any number but 0x29 and it must be under 0x7F. Going with 0x30 to 0x3F is probably OK.
    Keep sensor #1 awake, and now bring sensor #2 out of reset by setting its XSHUT pin high.
    Initialize sensor #2 with lox.begin(new_i2c_address) Pick any number but 0x29 and whatever you set the first sensor to
 */
void setID() {
  // all reset
  digitalWrite(SHT_LOX1, LOW);    
  digitalWrite(SHT_LOX2, LOW);
  delay(10);
  // all unreset
  digitalWrite(SHT_LOX1, HIGH);
  digitalWrite(SHT_LOX2, HIGH);
  delay(10);

  // activating LOX1 and reseting LOX2
  digitalWrite(SHT_LOX1, HIGH);
  digitalWrite(SHT_LOX2, LOW);

  // initing LOX1
  if(!lox1.begin(LOX1_ADDRESS)) {
    Serial.println(F("Failed to boot first VL53L0X"));
    while(1);
  }
  delay(10);

  // activating LOX2
  digitalWrite(SHT_LOX2, HIGH);
  delay(10);

  //initing LOX2
  if(!lox2.begin(LOX2_ADDRESS)) {
    Serial.println(F("Failed to boot second VL53L0X"));
    while(1);
  }
}

void read_dual_sensors() {
  
  lox1.rangingTest(&measure1, false); // pass in 'true' to get debug data printout!
  lox2.rangingTest(&measure2, false); // pass in 'true' to get debug data printout!

  // print sensor one reading
  Serial.print("1: ");
  if(measure1.RangeStatus != 4) {     // if not out of range
    Serial.print(measure1.RangeMilliMeter);
  } else {
    Serial.print("Out of range");
  }
  
  Serial.print(" ");

  // print sensor two reading
  Serial.print("2: ");
  if(measure2.RangeStatus != 4) {
    Serial.print(measure2.RangeMilliMeter);
  } else {
    Serial.print("Out of range");
  }
  
  Serial.println();
}

void setup() {
  Serial.begin(115200);

  // wait until serial port opens for native USB devices
  while (! Serial) { delay(1); }

  pinMode(SHT_LOX1, OUTPUT);
  pinMode(SHT_LOX2, OUTPUT);

  Serial.println("Shutdown pins inited...");

  digitalWrite(SHT_LOX1, LOW);
  digitalWrite(SHT_LOX2, LOW);

  Serial.println("Both in reset mode...(pins are low)");
  
  
  Serial.println("Starting...");
  setID();
 
}

void loop() {
   
  read_dual_sensors();
  delay(100);
}`

But you know, I probably couldn't execute a CTRL+A+CTRL+C CTRL+V correctly, so don't take my word for it, have a screencap.
[image]

chetan-prime, 2019-04-13:
The issue is in your 'vl53l0x_platform.h' . It seems to be corrupt as the one on github has none of the extra comments as the one on your file. I would suggest downloading from the entire module afresh from github the entire source as a zip file or using the git command and deleting whatever is there on your hard disk right now as its corrupt.