adafruit/Adafruit_VL53L0X #22: Library unable to work on ESP32

William-An, 2019-04-14:
# Platform
ESP32: Adafruit HUZZAH32 – ESP32 Feather Board ([Firmware Release 1.0.2](https://github.com/espressif/arduino-esp32/releases/tag/1.0.1))
VL53L0X: CJVL53L0XV2

# Problem
After burning example code into ESP32, serial prompts the following content and the board keep rebooting:

```
⸮)⸮]⸮!⸮⸮O⸮⸮k⸮⸮⸮⸮Յ⸮⸮!⸮Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception` was unhandled.
Core 1 register dump:
PC      : 0x400d5668  PS      : 0x00060130  A0      : 0x800d0f80  A1      : 0x3ffb1f10  
A2      : 0x3ffbdf10  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000017  
A6      : 0x00000000  A7      : 0x3ffbdf10  A8      : 0x800d5668  A9      : 0x3ffb1ee0  
A10     : 0x04c4b400  A11     : 0x00000016  A12     : 0x00000000  A13     : 0x00000064  
A14     : 0x00000000  A15     : 0x60013000  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400d5668:0x3ffb1f10 0x400d0f7d:0x3ffb1f30 0x400d48f6:0x3ffb1f50 0x400d0d81:0x3ffb1f80 0x400d6dab:0x3ffb1fb0 0x400877c1:0x3ffb1fd0

Rebooting...
⸮⸮⸮8⸮
AB⸮⸮⸮t!y⸮⸮:S⸮⸮⸮ĥ⸮Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d5668  PS      : 0x00060130  A0      : 0x800d0f80  A1      : 0x3ffb1f10  
A2      : 0x3ffbdf10  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000017  
A6      : 0x00000000  A7      : 0x3ffbdf10  A8      : 0x800d5668  A9      : 0x3ffb1ee0  
A10     : 0x04c4b400  A11     : 0x00000016  A12     : 0x00000000  A13     : 0x00000064  
A14     : 0x00000000  A15     : 0x60013000  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400d5668:0x3ffb1f10 0x400d0f7d:0x3ffb1f30 0x400d48f6:0x3ffb1f50 0x400d0d81:0x3ffb1f80 0x400d6dab:0x3ffb1fb0 0x400877c1:0x3ffb1fd0

Rebooting...`

```

## IDE Output
```
In file included from .../Arduino/libraries/Adafruit_VL53L0X/src/vl53l0x_device.h:37:0,
                 from .../Arduino/libraries/Adafruit_VL53L0X/src/vl53l0x_def.h:90,
                 from .../Arduino/libraries/Adafruit_VL53L0X/src/platform/src/vl53l0x_i2c_comms.cpp:2:
.../Arduino/libraries/Adafruit_VL53L0X/src/vl53l0x_types.h:56:108: note: #pragma message: Please review  type definition of STDINT define for your platform and add to list above 
  #pragma message("Please review  type definition of STDINT define for your platform and add to list above ")
```

leoncoolmoon, 2021-05-12:
@William-An have you sort out the issue? I have similar issue.
`08:06:22.982 -> Rebooting...
08:06:23.028 -> ets Jun  8 2016 00:22:57
08:06:23.028 -> 
08:06:23.028 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
08:06:23.028 -> configsip: 0, SPIWP:0xee
08:06:23.028 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
08:06:23.028 -> mode:DIO, clock div:1
08:06:23.028 -> load:0x3fff0018,len:4
08:06:23.028 -> load:0x3fff001c,len:1216
08:06:23.028 -> ho 0 tail 12 room 4
08:06:23.028 -> load:0x40078000,len:9720
08:06:23.028 -> ho 0 tail 12 room 4
08:06:23.028 -> load:0x40080400,len:6352
08:06:23.028 -> entry 0x400806b8
**08:06:24.763 -> SSD1306 started**
**08:06:24.763 -> VL53L0X pre begin**
08:06:24.763 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
08:06:24.763 -> Core 1 register dump:
08:06:24.763 -> PC      : 0x40166f60  PS      : 0x00060130  A0      : 0x800e2774  A1      : 0x3ffb1bc0  
08:06:24.810 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000034  A5      : 0x00000000  
08:06:24.810 -> A6      : 0xffffffff  A7      : 0x00000000  A8      : 0x00000100  A9      : 0x00000001  
08:06:24.810 -> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
08:06:24.810 -> A14     : 0x00060520  A15     : 0x00000000  SAR     : 0x00000005  EXCCAUSE: 0x0000001d  
08:06:24.810 -> EXCVADDR: 0x00000126  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
08:06:24.810 -> 
08:06:24.810 -> Backtrace: 0x40166f60:0x3ffb1bc0 0x400e2771:0x3ffb1be0 0x400e2806:0x3ffb1c00 0x400e2943:0x3ffb1c20 0x400dffff:0x3ffb1c50 0x400df849:0x3ffb1cc0 0x400d2fa1:0x3ffb1d00 0x400e7597:0x3ffb1fb0 0x4008e609:0x3ffb1fd0
08:06:24.856 -> 
08:06:24.856 -> Rebooting...
`

`void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
  Serial.begin(115200);
  i2c.begin(I2C_SDA, I2C_SCL);
  oled = Adafruit_SSD1306(SCREEN_WIDTH, SCREEN_HEIGHT, &i2c);
   if(!oled.begin(SSD1306_SWITCHCAPVCC, 0x3C)) { // Address 0x3C for 128x32
    Serial.println(F("SSD1306 allocation failed"));
    for(;;); // Don't proceed, loop forever
  }
 **Serial.println("SSD1306 started");**

  lox = Adafruit_VL53L0X();
  **Serial.println("VL53L0X pre begin");**

  if (!lox.begin(0x29, false, &i2c)) { //?
    Serial.println(F("Failed to boot VL53L0X"));
    for(;;); // Don't proceed, loop forever
  }
  **Serial.println("VL53L0X started");**
...
`

caternuson, 2021-12-15:
Could not reproduce this. Running library example:
https://github.com/adafruit/Adafruit_VL53L0X/blob/master/examples/vl53l0x/vl53l0x.ino
on a Feather ESP32 runs as expected:
[image]

Make sure Library and ESP32 Board Support Package are all updated to latest. Also, for testing, try running the same example sketch. Also remove anything else attached, like SSD1306 displays, etc.

Test just ESP32+VL53L0X.

closed 2021-12-28 by ladyada