adafruit/Adafruit_VL53L0X #23: No timeout implemented in code

Daybreakerflint, 2019-05-03:


PassiSchoppi, 2022-03-15:
I have the same issue. Did you manage to fix it somehow?

caternuson, 2022-03-15:
Can you provide complete example code that can be used to demonstrate the issue?

PassiSchoppi, 2022-03-15:
The example code examples/vl53l0x/vl53l0x.ino with one simple sensor connected as described in [this tutorial](https://cdn-learn.adafruit.com/downloads/pdf/adafruit-vl53l0x-micro-lidar-distance-sensor-breakout.pdf). The issue first happened to me after smacking the sensor on the table… I think you can replicate it more consistently by just pulling it off and then plugging it back in… The issue is not really with the sensor but how the API reacts. For me the whole program on the Arduino just stops completely. I would rather like to have it so that the API tells me that the sensor wasn‘t reachable so that i can either try to reconnect by resetting all i2c addresses or by just escaping the sensor value to be like a default or something. Does that make sense? If you want to I can try to send you a video demonstrating the issue.

closed 2022-03-15 by caternuson