adafruit/Adafruit_VL53L0X #24: Library does not work when external power source is used

markg85, 2019-05-18:
Hi,

I know this sounds weird, but it really appears to be the case here.
I can connect other sensors just fine (the TSL2591 for example), on an **external power source** too, and it just works. Note: i want to use an external power source because feeding the i2c censors i have via just arduino is simply going to draw too much current.
When i swap the wires from the TSL2591 to the VL53L0X, it simply fails with:

> Adafruit VL53L0X test
> Failed to boot VL53L0X

And yes, if i use the arduino (micro) to power the chip, it works just fine. I do have a common ground.

You probably need more information so i enabled debugging:
```
void setup() {
  Serial.begin(115200);

  // wait until serial port opens for native USB devices
  while (! Serial) {
    delay(1);
  }
  
  Serial.println("Adafruit VL53L0X test");
  if (!lox.begin(0x29, true)) {
    Serial.println(F("Failed to boot VL53L0X"));
    while(1);
  }
  // power 
  Serial.println(F("VL53L0X API Simple Ranging example\n\n")); 
}
```

Note the:
>   if (!lox.begin(0x29, true)) {

With that enabled, it spits out:

> Adafruit VL53L0X test
> VL53L0X Info:
> Device Name: VL53L0X TS1, Type: VL53L0X, ID: ␃A`p8␜␎␇␃A`p8␜␎␇␃A
> Rev Major: 1, Minor: 1
> VL53L0X: StaticInit
> VL53L0X Error: -50
> Failed to boot VL53L0X

The error (-50) means:
> 	/*!< In the function a division by zero occurs */
> #define VL53L0X_ERROR_REF_SPAD_INIT			((VL53L0X_Error) -50)

Which clearly points at a code issue... Somewhere...
When further investigating that error, i found this thread:
https://community.st.com/s/question/0D50X00009XkYOCSA3/spad-initialization-error

Which has a response (from just one month ago) from a ST employee with an apparent fix. I'm not quite sure if that is "the" fix for this issue, i haven't testes it yet.

Is there any more information that is required of me to fix this?

Best regards,
Mark

caternuson, 2019-05-23:
This sounds more like a hardware issue. Please post in the forums with photos of your setup showing how everything is connected:
https://forums.adafruit.com/

markg85, 2019-05-23:
> This sounds more like a hardware issue. Please post in the forums with photos of your setup showing how everything is connected:
> https://forums.adafruit.com/

I doubt that will help.

Fun fact, it does work when hooking it up to an ESP8266 with the very same library and an external power supply. I do have to change the begin(...) function there though as the i2c pins are whatever i choose but i must choose as there is no hardware i2c.

It doesn't work on the arduino nano with **external** power. Before i go on and post on the forums, could you **verify** that the same chip does work in the same setup? (that is on the Arduino nano)

caternuson, 2019-05-23:
We can check that also in the forums.

mgulsoy, 2019-07-27:
The error you get is not because of a hardware issue. The sensor has completed static_init. But after static_init the calibration fails. 

`VL53L0X_ERROR_REF_SPAD_INIT	((VL53L0X_Error) -50)`

is happening because possibly the factory calibration data in nvm (of the sensor) is corrupt. Skipping the calibration step may cause the sensor to work. Did you try other libs?

markg85, 2019-07-27:
It's not an issue for me anymore.
I was having this - repeatedly - on the arduino micro when the VL53L0X was externally powered. For me, that was merely a test setup to get it working. My final setup would be with a wifi enabled chip (the ESP8266 in this case) which works just fine.

@caternuson i'm not going to do that. It very clearly is a bug. I get console output which makes it a software thing, not a wiring issue. So i ultimately would end up here anyhow.

As i just said, the issue - for me - is just not there on the ESP8266. That doesn't mean that this issue isn't there. Just that i'm apparently not triggering it on this setup which did trigger on the arduino micro.

caternuson, 2019-08-16:
Closing this. Sounds like a hardware issue. Worked for a ESP8266. Did not work on a Arduino Micro. Request to post in forums with additional details not provided, so difficult to troubleshoot. Also:

> It's not an issue for me anymore.

Can re-open if this comes up again.

closed 2019-08-16 by caternuson

TanveerHussain123, 2020-11-16:
i solve the same issue in my project when use external power face the issue of sensor detection.
you can watch the video to solve the issue. 
https://youtu.be/QEiNbTB8nNE
one more recommendation beside the video procedure add a capacitor near to sensor.