adafruit/Adafruit_VL53L0X #25: Adafruit_VL53L0X.cpp file possible typo

mgulsoy, 2019-07-17:
In the file Adafruit_VL53L0X.cpp at line 100 it is written:

`if( ( DeviceInfo.ProductRevisionMinor != 1 ) && ( DeviceInfo.ProductRevisionMinor != 1 ) )`

I think this line should be:

`if( ( DeviceInfo.ProductRevisionMajor != 1 ) && ( DeviceInfo.ProductRevisionMinor != 1 ) )`

ladyada, 2019-07-17:
good question, does it still work for you if you change that line :)

mgulsoy, 2019-07-21:
Yes, it works but with the newer v1.15 chips, this line could be an issue.

ladyada, 2019-07-21:
ok then please remove the minor revision check and submit a PR, it doesnt make sense to fix a typo that then breaks the code

mgulsoy, 2019-07-22:
Ok :) ( By the way, I admire your work @ladyada )

closed 2019-07-22 by mgulsoy

ladyada, 2019-07-22:
thanks for the kind note! :)