adafruit/Adafruit_VL53L0X #27: Question: How to disable Informations?

odo2063, 2019-10-09:
Hello!
How may i disable all this Information when i just want the last line?

`       Writing 1 to addr 0x80: 0x1, 
	Writing 1 to addr 0xFF: 0x1, 
	Writing 1 to addr 0x0: 0x0, 
	Writing 1 to addr 0x91: 0x3C, 
	Writing 1 to addr 0x0: 0x1, 
	Writing 1 to addr 0xFF: 0x0, 
	Writing 1 to addr 0x80: 0x0, 
	Writing 1 to addr 0x0: 0x1, 
	Reading 1 from addr 0x0: 0x0, 
	Reading 1 from addr 0x13: 0x40, 
	Reading 1 from addr 0x13: 0x40, 
	Reading 1 from addr 0x13: 0x40, 
	Reading 1 from addr 0x13: 0x40, 
	Reading 1 from addr 0x13: 0x40, 
	Reading 1 from addr 0x13: 0x40, 
	Reading 1 from addr 0x13: 0x40, 
	Reading 1 from addr 0x13: 0x40, 
	Reading 1 from addr 0x13: 0x40, 
	Reading 1 from addr 0x13: 0x40, 
	Reading 1 from addr 0x13: 0x44, 
	Reading 12 from addr 0x14: 0x5B, 0x6, 0x4C, 0x4, 0x33, 0xE3, 0xA, 0xF9, 0x0, 0x12, 0x0, 0xA0, 
	Writing 1 to addr 0xFF: 0x1, 
	Reading 2 from addr 0xB6: 0x9, 0x58, 
	Writing 1 to addr 0xFF: 0x0, 
	Writing 1 to addr 0xB: 0x1, 
	Writing 1 to addr 0xB: 0x0, 
	Reading 1 from addr 0x13: 0x40, 
160
`

My code looks like this:
`    lox.getSingleRangingMeasurement(&measure, false);
    Serial.println(measure.RangeMilliMeter);`

caternuson, 2019-10-09:
Somewhere in your code you must be `#define`-ing `I2C_DEBUG`:
https://github.com/adafruit/Adafruit_VL53L0X/blob/master/src/platform/src/vl53l0x_i2c_comms.cpp#L14
So just don't do that define.

It should be off by default:
https://github.com/adafruit/Adafruit_VL53L0X/blob/master/src/platform/src/vl53l0x_i2c_comms.cpp#L4

odo2063, 2019-10-10:
öhm...i don't do that...It is mostly the example code...

`#include "Adafruit_VL53L0X.h"

Adafruit_VL53L0X lox = Adafruit_VL53L0X();
VL53L0X_RangingMeasurementData_t measure;

void setup() {
  Serial.begin(115200);

  // wait until serial port opens for native USB devices
  while (! Serial) {
    delay(1);
  }
  
  Serial.println("Adafruit VL53L0X test");
  if (!lox.begin()) {
    Serial.println(F("Failed to boot VL53L0X"));
    while(1);
  }
  // power 
  Serial.println(F("VL53L0X API Simple Ranging example\n\n")); 
}

void loop() {
  //VL53L0X_RangingMeasurementData_t measure;
    
  //Serial.print("Reading a measurement... ");
  //lox.rangingTest(&measure, false); // pass in 'true' to get debug data printout!
  
  //if (measure.RangeStatus != 4) {  // phase failures have incorrect data
    //Serial.print("Distance (mm): "); 
    lox.getSingleRangingMeasurement(&measure, false);
    Serial.println(measure.RangeMilliMeter);
  //} else {
  //  Serial.println(" out of range ");
  //}
    
  delay(100);
}`

caternuson, 2019-10-10:
If I run the sketch above I get this in the serial console:
[image]
I'm not seeing the debug output.

How did you install the library?

odo2063, 2019-10-10:
the library manager...so I guess the only difference might be the board which is a STM NUCLEO in my case...
what happens if you use my sketch?

caternuson, 2019-10-10:
The serial console above was from running your sketch - the one you posted above.

It *may* be board related. Can you try another board? Or try grepping through the STM NUCELO board support files and see if somewhere in there `I2C_DEBUG` is being `#define`d.

odo2063, 2019-10-11:
hm...It's #defined in libraries/Adafruit_VL53L0X/src/platform/src/vl53l0x_i2c_comms.cpp

caternuson, 2019-10-11:
Yes, but it should be commented out by default:
https://github.com/adafruit/Adafruit_VL53L0X/blob/master/src/platform/src/vl53l0x_i2c_comms.cpp#L4
Is that line not commented out in your copy of the library?

ladyada, 2019-10-11:
hiya it could be somethign in your arduino definition for the nucleo defining I2C_DEBUG. there's nothing we can do about it here!

closed 2019-10-11 by ladyada