adafruit/Adafruit_VL53L0X #28: Reopening: https://github.com/adafruit/Adafruit_VL53L0X/issues/27

odo2063, 2019-10-13:
@caternuson: I installed it with the arduino library manager, and did not modfify it. It seems there is/was(?) a problem with the way of installing. Maybe it's got some old or debug branch.

caternuson, 2019-10-14:
Are you re-opening because the problem still exists? If you compare the file that is in your library folder with the repo, is it different? Is the line the #defines I2C_DEBUG uncomented? Have you searched through the files associated with the STM NUCLEO Board Support Package to make sure there is not a #define buried in there somewhere? Do you have an Arduino UNO, or something other than an STM NUCLEO, to try and see if you get the same issue?

caternuson, 2019-10-14:
Run this sketch and post what you get in the serial console:
```cpp
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("Scooby-Doo");
#ifdef I2C_DEBUG
  Serial.println(", where are you?");
#endif
  delay(1000);
}
```

odo2063, 2019-10-14:
i commented it out in the adafruit library, where it was defined. works now.
but the question is what could happend so it was not commented out in the lib?

caternuson, 2019-10-14:
The question is why it is being installed with the line uncommented. I can not recreate this. If I install via Library Manager in Arduino, the line is commented out.

odo2063, 2019-10-14:
what os do you use?

caternuson, 2019-10-14:
Linux. But the Library Manager shouldn't have any OS specific behavior.

odo2063, 2019-10-14:
me too..so I tried some downgrading/upgrading of the lib. now it stays commented out... :-/  strange....

caternuson, 2019-10-14:
It could be something with local caching. Not sure. But really sounds like something with your Arduino setup. Just make sure the line is commented out. If you can figure out exactly why/how you are getting it installed with the line uncommented, you can reopen this issue.

closed 2019-10-14 by caternuson

odo2063, 2019-10-14:
ok...so thanks for the help...