adafruit/Adafruit_VL53L0X #30: Reference and crosstalk calibration routines givins strange results

Gregorino, 2019-12-22:
Routine `VL53L0X_perform_offset_calibration` from src/core/src/vl53l0x_api_calibration.cpp is giving strange results. By my opinion the problem is in `uint16_t sum_ranging` which should be of type 
`uint32_t ` due to later bit shifting (<<16), which needs to be done due to fixed point arithmetics.
(I was performing this calibration at 270 mm distance, which is due to documentation not optimal one, but after the fixes it works fine).

In case of `VL53L0X_perform_xtalk_calibration` the same goes for multiple variables (sum_ranging, ...) that accumulate the range measurements performed by the routine. Unfortunately there are some other errors present due to which the calibration routine doesn't produce an appropriate `XTalkCompensationRateMegaCps` value. Due to the lack of understanding what this calibration routine is trying to compute, I am not able to figure it out why.
(I was performing this calibraton at 600 mm distance).

ladyada, 2019-12-22:
hi that code is from STM, we didn't change any of it - so you'd have to ask ST for advice or guidance!

closed 2019-12-22 by ladyada

Gregorino, 2019-12-22:
I know, but it still affects you library. Anyway, I have already asked ST and will also let you know when I get the answer. If the problem is in the code and not in me using it, I can also apply the fixes in your repo.

ladyada, 2019-12-22:
awesome, yes please do that would be great :)