adafruit/Adafruit_VL53L0X #35: Doxygen index page is blank

dherrada, 2020-05-06:


siddacious, 2020-06-06:
After taking a look, this will likely require a change to the Doxygen config because all the code is within `src/`

dherrada, 2020-06-08:
@siddacious how would you reccomend I go about doing this? Do you think we should have the master Doxyfile check in src/ in addition to where it's already checking or should we have this repository and any others that have src/ their own special Doxyfile?

siddacious, 2020-06-08:
TBH I'm not sure if it would make sense to flatten the code a bit and pull our code out of `src/` and into the root, or changing the Doxyfile. I'm not familiar enough with the expected or permissible layouts, but from my experience `src/` are uncommon.

@ladyada any input?

ladyada, 2020-06-08:
you could try to move the .cpp/.h files down to root level - i'd do that in a branch to verify everything still compiles, and use git mv so you dont lose history

dherrada, 2020-06-08:
@ladyada would that break libraries that have this as a dependancy?

ladyada, 2020-06-08:
no, but you may have to adjust the #include's for the new paths

dherrada, 2020-06-08:
Ok

closed 2020-07-10 by dherrada