adafruit/Adafruit_VL53L0X #55: Need missing dox from PR #49

caternuson, 2022-03-16:
PR #49 *was* passing CI checks on its last commit. But after merging, CI is failing on missing dox. Not sure why that wasn't caught in the PR itself. Maybe the CI checks changed.

Anywho, need dox for:
```
/home/runner/work/Adafruit_VL53L0X/Adafruit_VL53L0X/src/Adafruit_VL53L0X.h:84: warning: Member startMeasurement(boolean debug=false) (function) of class Adafruit_VL53L0X is not documented.
/home/runner/work/Adafruit_VL53L0X/Adafruit_VL53L0X/src/Adafruit_VL53L0X.h:85: warning: Member stopMeasurement(boolean debug=false) (function) of class Adafruit_VL53L0X is not documented.
/home/runner/work/Adafruit_VL53L0X/Adafruit_VL53L0X/src/Adafruit_VL53L0X.h:86: warning: Member getLimitCheckCurrent(uint8_t LimitCheckId, FixPoint1616_t *pLimitCheckCurrent, boolean debug=false) (function) of class Adafruit_VL53L0X is not documented.
/home/runner/work/Adafruit_VL53L0X/Adafruit_VL53L0X/src/Adafruit_VL53L0X.h:89: warning: Member getDeviceMode(VL53L0X_DeviceModes *pDeviceMode, boolean debug=false) (function) of class Adafruit_VL53L0X is not documented.
/home/runner/work/Adafruit_VL53L0X/Adafruit_VL53L0X/src/Adafruit_VL53L0X.h:91: warning: Member setDeviceMode(VL53L0X_DeviceModes DeviceMode, boolean debug=false) (function) of class Adafruit_VL53L0X is not documented.
/home/runner/work/Adafruit_VL53L0X/Adafruit_VL53L0X/src/Adafruit_VL53L0X.h:94: warning: Member setInterruptThresholds(FixPoint1616_t ThresholdLow, FixPoint1616_t ThresholdHigh, boolean debug=false) (function) of class Adafruit_VL53L0X is not documented.
/home/runner/work/Adafruit_VL53L0X/Adafruit_VL53L0X/src/Adafruit_VL53L0X.h:97: warning: Member getInterruptThresholds(FixPoint1616_t *pThresholdLow, FixPoint1616_t *pThresholdHigh, boolean debug=false) (function) of class Adafruit_VL53L0X is not documented.
/home/runner/work/Adafruit_VL53L0X/Adafruit_VL53L0X/src/Adafruit_VL53L0X.h:100: warning: Member clearInterruptMask(boolean debug=false) (function) of class Adafruit_VL53L0X is not documented.
/home/runner/work/Adafruit_VL53L0X/Adafruit_VL53L0X/src/Adafruit_VL53L0X.h:102: warning: Member getGpioConfig(VL53L0X_DeviceModes *pDeviceMode, VL53L0X_GpioFunctionality *pFunctionality, VL53L0X_InterruptPolarity *pPolarity, boolean debug=false) (function) of class Adafruit_VL53L0X is not documented.
/home/runner/work/Adafruit_VL53L0X/Adafruit_VL53L0X/src/Adafruit_VL53L0X.h:106: warning: Member setGpioConfig(VL53L0X_DeviceModes DeviceMode, VL53L0X_GpioFunctionality Functionality, VL53L0X_InterruptPolarity Polarity, boolean debug=false) (function) of class Adafruit_VL53L0X is not documented.
```

@Vgdn1942 Can you make a follow on PR with dox?

caternuson, 2022-03-16:
Some general information here about doxygen:
https://learn.adafruit.com/the-well-automated-arduino-library/doxygen

Just need to add specially formatted comment block before each method. Can use some of the other ones for reference. For example:
https://github.com/adafruit/Adafruit_VL53L0X/blob/9a3bef2d8c5195d19e153629874f9c23e2d0ad1c/src/Adafruit_VL53L0X.cpp#L208-L222

caternuson, 2022-03-16:
This was your pull request?
https://github.com/adafruit/Adafruit_VL53L0X/pull/49

Vgdn1942, 2022-03-16:
> This was your pull request? #49

Yes, but it lists the author of the original commit.  I just tweaked the style of the code that was pointed out there.

caternuson, 2022-03-16:
OK, nvm. I'll PR it.

caternuson, 2022-03-17:
Fixed with #56

closed 2022-03-17 by caternuson