adafruit/Adafruit_VL53L0X #57: Annoying Reading After Out of Range

matriawan, 2022-08-14:
I just use example code and everything is work fine, but when I put the sensor without a barrier or to get out of range, the output from Arduino Serial Monitoring shows random numbers such as 2mm or 4mm. How to solve this, just make sure the output always out of range not random numbers.

fafiunal, 2022-08-31:
i am having the same problems. I use a protective glass as an extra. I occasionally read absurd values when the sensor is out of range.

caternuson, 2022-08-31:
Which example code is being run? Is `RangeStatus` being checked?

fafiunal, 2022-09-01:
I use interrupt example with NEW_MEASURE_READY. I am checking RangeStatus variable, when i measure distance. 

When I occasionally read absurd values, ,I observe that range status is 2(Signal Fail). @caternuson @matriawan

caternuson, 2022-09-01:
> When I occasionally read absurd values, ,I observe that range status is 2(Signal Fail).

Only occasionally? Sometimes you get absurd values with a range status that is OK?

fafiunal, 2022-09-04:
Sorry for my late response.

Yes occasionally.

 I get absurd values with range status that is "2" or "4". 
You can say that you can read measurement when a range status is "0". But, After 200mm, range status is changed from "0 " to "2" by sensor. 200mm measured distance is not enough for me. :) @caternuson @matriawan

caternuson, 2022-09-06:
The example sketches may not fully cover all possible failure cases. For example, the interrupt example only checks for phase failures:
https://github.com/adafruit/Adafruit_VL53L0X/blob/e71010b98c862b9bc39bad79f801bd2522eec99d/examples/vl53l0x_Interrupt/vl53l0x_Interrupt.ino#L78

In general, if the status is anything other than valid, then the measurement should be considered incorrect. Various things can cause this. The status code helps indicate why.

2 = Signal Fail
4 = Phase Fail

@fafiunal This could be related to your use of cover glass.

@matriawan Are you using a cover glass of some kind? What is the target object? What are the ambient lighting conditions?

viktak, 2024-05-21:
@fafiunal 
Having a behavior here too...:
In my case, even when the status code is 2, the measured distance is still correct. However, the device seems to be measuring distances only up to ~ 1300mm
Have you figured how to fix/work around this?

caternuson, 2024-05-21:
A status code of 2 is a signal fail, so the reading would not be correct in that case. Only a status code of 0 is valid.

https://github.com/adafruit/Adafruit_VL53L0X/blob/23dc309c56ae7b3031b296380b0663c5dd5edb68/src/core/src/vl53l0x_api_strings.cpp#L185-L218

viktak, 2024-05-24:
@caternuson I understand the concept :) but experiments show, that as long as the distance is under 1300mm, the measurement is correct, even if the status code is 2 (300-1300mm).
Could it be because of a (faulty) sensor? I'll try it later...

caternuson, 2024-05-24:
It could be for many reasons. Target size. Target color. Ambient lighting conditions. Sensor settings. etc.

Keep in mind the sensor has a field of view of 25deg on the range sensing element:
[image]
That equates to an object size of over 500mm at a distance of 1300mm, and over 800mm at the 2000mm full range.

In terms of sensor setting, these are the options:
https://github.com/adafruit/Adafruit_VL53L0X/blob/23dc309c56ae7b3031b296380b0663c5dd5edb68/src/Adafruit_VL53L0X.h#L44-L50
which have different trade offs in performance:
[image]
Only the long range mode provides sensing beyond 1200mm, and has caveats of being in dark conditions without any infrared (IR) sources.

It is probably worth reading through the datasheet and other documents to verify the sensor is being configured and used correctly for your application:
https://www.st.com/en/imaging-and-photonics-solutions/vl53l0x.html#documentation

In general, these issue do not sound like library code issues.

viktak, 2024-05-24:
@caternuson Thank you for all the info you mention, I appreciate it!
I didn't think for a moment it was a library issue (I checked the source code before coming here) - I only came here to see if others have encountered the same thing.
Thanks again for your efforts!!