adafruit/Adafruit_VL53L0X #60: Interrupt Example don't work

cablesky, 2022-12-18:
The interrupt example doesn't work. The interrupt is simply ignored. I.e. the distance is displayed although only the range of 50-100mm should be displayed.

caternuson, 2022-12-18:
The description is confusing. Can you post the output you are seeing in the Serial Monitor when running the example.

cablesky, 2022-12-18:
Set GPIO Config so if range is lower the LowThreshold trigger Gpio Pin 
Set Interrupt Threasholds... 
sVL53L0X: setInterruptThresholds
Set Mode VL53L0X_DEVICEMODE_CONTINUOUS_RANGING... 
StartMeasurement... 
Reading a measurement... Distance (mm): 55
Reading a measurement... Distance (mm): 55
Reading a measurement... Distance (mm): 55
Reading a measurement... Distance (mm): 55
Reading a measurement... Distance (mm): 57
Reading a measurement... Distance (mm): 57
Reading a measurement... Distance (mm): 57
Reading a measurement... Distance (mm): 57
Reading a measurement... Distance (mm): 57
Reading a measurement... Distance (mm): 57
Reading a measurement... Distance (mm): 58
Reading a measurement... Distance (mm): 58
Reading a measurement... Distance (mm): 58
Reading a measurement... Distance (mm): 58
Reading a measurement... Distance (mm): 58
Reading a measurement... Distance (mm): 55
Reading a measurement... Distance (mm): 55
Reading a measurement... Distance (mm): 55
Reading a measurement... Distance (mm): 55
Reading a measurement... Distance (mm): 55

caternuson, 2022-12-18:
Thanks. The output looks generally as expected. Is the issue that the distance value is not changing?

Does the non-interrupt example work OK on the same setup?
https://github.com/adafruit/Adafruit_VL53L0X/blob/master/examples/vl53l0x/vl53l0x.ino

cablesky, 2022-12-19:
Thank you for your quick answer!
Yes, the example without an interrupt works perfectly.
The example with interrupt also outputs the distance values. Only the interrupt is not triggered.

caternuson, 2022-12-19:
The interrupt must be triggering, otherwise there would not be any output. The output is surrounded by a conditional that will only occur if the interrupt has fired.

The behavior of the interrupt with respect to the high/low limits is configurable.
[image]

The example sketch configures it to trigger when Level Low:
```cpp
  lox.setGpioConfig(VL53L0X_DEVICEMODE_CONTINUOUS_RANGING,
                    VL53L0X_GPIOFUNCTIONALITY_THRESHOLD_CROSSED_LOW,
                    VL53L0X_INTERRUPTPOLARITY_LOW);
```

To make sure this is working, tested here and ran the example sketch on a QT PY M0 and get this output:
```
VL53L0X API Interrupt Ranging example

Set GPIO Config so if range is lower the LowThreshold trigger Gpio Pin 
Set Interrupt Threasholds... 
sVL53L0X: setInterruptThresholds
Set Mode VL53L0X_DEVICEMODE_CONTINUOUS_RANGING... 
StartMeasurement... 
Reading a measurement... Distance (mm): 48
Reading a measurement... Distance (mm): 50
Reading a measurement... Distance (mm): 49
Reading a measurement... Distance (mm): 46
Reading a measurement... Distance (mm): 46
Reading a measurement... Distance (mm): 46
Reading a measurement... Distance (mm): 44
Reading a measurement... Distance (mm): 41
Reading a measurement... Distance (mm): 45
Reading a measurement... Distance (mm): 45
Reading a measurement... Distance (mm): 39
Reading a measurement... Distance (mm): 41
Reading a measurement... Distance (mm): 39
Reading a measurement... Distance (mm): 39
```

Note none of the values are above 50mm.

You are getting values above 50mm in your output. That would be an indication of false triggers. Not lack of trigger.

Double check your wiring and make sure the VL53L0X's GPIO pin is connected to the pin referenced in the sketch.

caternuson, 2022-12-20:
Please post in the forums with photos of your setup showing how everything is connected.
https://forums.adafruit.com/

awong1900, 2023-09-21:
> The interrupt must be triggering, otherwise there would not be any output. The output is surrounded by a conditional that will only occur if the interrupt has fired.
> 
> The behavior of the interrupt with respect to the high/low limits is configurable. [image]
> 
> The example sketch configures it to trigger when Level Low:
> 
> ```c++
>   lox.setGpioConfig(VL53L0X_DEVICEMODE_CONTINUOUS_RANGING,
>                     VL53L0X_GPIOFUNCTIONALITY_THRESHOLD_CROSSED_LOW,
>                     VL53L0X_INTERRUPTPOLARITY_LOW);
> ```
> 
> To make sure this is working, tested here and ran the example sketch on a QT PY M0 and get this output:
> 
> ```
> VL53L0X API Interrupt Ranging example
> 
> 
> Set GPIO Config so if range is lower the LowThreshold trigger Gpio Pin 
> Set Interrupt Threasholds... 
> sVL53L0X: setInterruptThresholds
> Set Mode VL53L0X_DEVICEMODE_CONTINUOUS_RANGING... 
> StartMeasurement... 
> Reading a measurement... Distance (mm): 48
> Reading a measurement... Distance (mm): 50
> Reading a measurement... Distance (mm): 49
> Reading a measurement... Distance (mm): 46
> Reading a measurement... Distance (mm): 46
> Reading a measurement... Distance (mm): 46
> Reading a measurement... Distance (mm): 44
> Reading a measurement... Distance (mm): 41
> Reading a measurement... Distance (mm): 45
> Reading a measurement... Distance (mm): 45
> Reading a measurement... Distance (mm): 39
> Reading a measurement... Distance (mm): 41
> Reading a measurement... Distance (mm): 39
> Reading a measurement... Distance (mm): 39
> ```
> 
> Note none of the values are above 50mm.
> 
> You are getting values above 50mm in your output. That would be an indication of false triggers. Not lack of trigger.
> 
> Double check your wiring and make sure the VL53L0X's GPIO pin is connected to the pin referenced in the sketch.

How to set trigger if i want get above 50mm?

disq, 2024-02-10:
The issue with the example is the high threshold is not used and it should trigger when it's the ranged value is BELOW the low threshold (5cm)... because `VL53L0X_GPIOFUNCTIONALITY_THRESHOLD_CROSSED_LOW` is set.

The API details and how the thresholds can be defined is explained in the [UM2039 API User Manual](https://www.st.com/resource/en/user_manual/um2039-world-smallest-timeofflight-ranging-and-gesture-detection-sensor-application-programming-interface-stmicroelectronics.pdf)

caternuson, 2024-02-11:
Is there an actual issue with the example? Or is this just confusion about the interrupt behavior?

matteogrisenti, 2024-02-18:
`#define INTERRUPT_PIN P1_5

Adafruit_VL53L0X sensor = Adafruit_VL53L0X();

void setup() {
  Serial.begin(115200);
  
  // wait until serial port opens for native USB devices
  while (! Serial) {
    delay(1);
  }

  pinMode(INTERRUPT_PIN, INPUT_PULLUP); // Configura il pin di interruzione come input
  attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN), interruptHandler, FALLING); // Associa l'ISR alla caduta dell'interruzione
  Wire.begin();

  if (!sensor.begin()) {
    Serial.println(F("Failed to boot VL53L0X"));
    while(1);
  }
  
  // Configura il sensore per abilitare l'interruzione su nuova misurazione
  sensor.setInterruptThresholds(0, 100); // Imposta soglie di interruzione
  sensor.startMeasurement();             // Avvia le misurazioni continue

  sensor.setGpioConfig( VL53L0X_DEVICEMODE_CONTINUOUS_RANGING,
                        VL53L0X_GPIOFUNCTIONALITY_THRESHOLD_CROSSED_LOW,
                        VL53L0X_INTERRUPTPOLARITY_LOW);

}

void loop() {
  VL53L0X_RangingMeasurementData_t measure;
  sensor.rangingTest(&measure);
  
  Serial.print("Reading a measurement... ");
  
  if (measure.RangeStatus != 4) { // phase failures have incorrect data
    Serial.print("Distance (mm): "); Serial.print(measure.RangeMilliMeter);
    Serial.print("Range: "); Serial.println(measure.RangeStatus);
  } else {
    Serial.println(" out of range ");
  }
  
  delay(1000);
}

void interruptHandler(){
  Serial.print("YAAAAAAAAAAAAAAAAAAAAAAAAAAAAAa ");
  sensor.clearInterruptMask(false);
}`

This is my code, I use a MSP432P401 board and coding on Energia IDE; if I comment the   attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN), interruptHandler, FALLING); line all work, if Itry to use interrupt uncomment this line i can't see anything on the serial console

closed 2026-02-09 by caternuson